Xinyuan-LilyGO/LilyGo-T-Relay #2: Is it possible to add external anten for Wifi  ?

David-2D3FE9, 2021-11-06:
Hello first of all, thank you for the versatily and the quality of your products.   
I am realy enjoy to discover your T-Relay board and I am wondering if it is possible to add an external antenna for the wifi.  

On the product picture it seem to have pads near ESP32,   
Is it for soldering an SMA connector for external Wifi Antenna ?   

[image]

Thanks in advance for your response

LilyGO, 2021-11-09:
Hello. Modules have onboard antennas. If external antennas are used, jump resistors and weld antenna seats are required.

David-2D3FE9, 2021-11-24:
Thanks for your response.
My delivery isdelayed so I will see that in January :-)

closed 2021-11-24 by David-2D3FE9