Xinyuan-LilyGO/LilyGo-T-Relay #3: What is the maximum current rating for the relay ?

X-Ryl669, 2021-11-19:
I see you have mounted 10A relay. Are the screw terminal block and PCB trace matched for 10A ?

LilyGO, 2021-11-20:
Yes. You can refer to the parameters of the relay

X-Ryl669, 2021-11-20:
The trace width seems like they are 3.8mm on the bottom layer but 7.6mm is required to pass 10A on a 1oz PCB (unless there is a trace going under the relay themselves on the top layer too).
Is the PCB copper layer made of 2oz ? 
Or is there a trace on the relay pin on the top layer too ?

LilyGO, 2021-11-22:
Yes, it is not recommended to exceed the limit value.

SublimeMudTime, 2022-03-04:
It almost looks like there are traces for the relay on the top and bottom when examining my board. And 2 X 3.8 = 7.6 so it seems ok...

X-Ryl669, 2022-03-04:
Thanks! I've received mine recently and indeed, there are traces on both side so it's ok.

closed 2022-03-04 by X-Ryl669