Xinyuan-LilyGO/LilyGo-T-Relay #9: Pin swap required on the connector

X-Ryl669, 2022-05-30:
Hi, 

For a future version, it would be better if you could swap the Vcc (3.3V) and GND on the pins row:
```
[... | 15 | 14 | Gnd | Gnd ]
[... | 04 | 02 | 3.3V| 3.3V]
``` 
to this instead
```
[... | 15 | 14 | 3.3V| Gnd ]
[... | 04 | 02 | 3.3V| Gnd ]
```

Because if you want to make independent connectors, you're more likely to use rows (15,14,Vcc,Gnd) and (4,2,Vcc,Gnd) than columns.

iwoodcock1952, 2022-11-28:
I have 4-relay item. IO21 is normally default I2C SDA..which pin is assigned I2C in this case?

X-Ryl669, 2022-11-28:
You choose what pin to use when instantiating the I2C driver. It's free to use any of them except 32-39 that are readonly IIRC.

iwoodcock1952, 2022-12-01:
Dear Sirs,

I am trying to use the LILYGO ESP32  2021-7-16 V1.0 Relay unit with ESP32
but I find it is almost impossible to program it.

Sometimes randomly it accepts a program, but then refuses to accept
revisions.

I am using Arduino 1.8.18 on latest updated Windows 10.

My aim is to use the dual core capability but it is very difficult when I
can not upload program updates.

Website is no help. Github no help.

It is probable I will discontinue the use of these devices in my projects,
unless you can provide a CLEAR and UNAMBIGUOUS guide how to upload programs
reliably.

KInds regards

Ian Woodcock

ZaleAnderson, 2022-12-09:
> 
Hey, make sure you have a usb to ttl adapter for the board, it does not come with an on board converter so you need an external one

iwoodcock1952, 2022-12-09:
Good morning
Yes thanks I have the connector.
What I have so far failed is to get the relay unit to detect any i2c
devices using suggestions I've seen on the internet.

Please provide your recommended code to assign i2c to pins 13 and 14 for
example

Kind regards

Ian Woodcock

jaseiffert, 2022-12-14:
> Dear Sirs, I am trying to use the LILYGO ESP32 2021-7-16 V1.0 Relay unit with ESP32 but I find it is almost impossible to program it. Sometimes randomly it accepts a program, but then refuses to [...]

I've had problems with my T-U2T Adapter and had intermittent problems till I pushed the USB C cable into the T-U2T very far till I heard a click. It worked much better after that. I'm not sure why but there is barely any silver of the connector showing.

Thanks,

Jeffery

jaseiffert, 2022-12-14:
> Good morning Yes thanks I have the connector. What I have so far failed is to get the relay unit to detect any i2c devices using suggestions I've seen on the internet. Please provide your recommended code to assign i2c to pins 13 and 14 for example Kind regards Ian Woodcock
>

Depending on your I2C device you can use the following.

#include <Wire.h>

#define I2C_SDA 33
#define I2C_SCL 32
// set the pins you want to use

Under Void setup()
Wire.begin(I2C_SDA, I2C_SCL);

Some libraries you don't need the Wire.begin, just add the I2C_SDA and I2C_SCL to the libraries begin or init. You'll have to read how to do that for different I2C device libraries.

Random Nerd Tutorials is a great source for ESP32 Example and Programming help.

Thanks,

Jeffery

jaseiffert, 2022-12-14:
I'm also disappointed that the default I2C and SPI pins are used for the relays. 

SPI can be used for displays.

If you ever make V2 revision of the board leave the I2C and SPI pins alone and use something else. 

This board really needs an external Antenna connector for the ESP32.

Instead of using IO25 for a led, just connect the led to the 3v3 voltage coming out of the regulator as just a power indicator.

You could also use a PCF8574 I2C GPIO Expander to control the relays and leave the ESP32 ports available for other things but I'm no Electronics Engineer so I have no clue how involved it is. I've just seen others use this chip to control relays or other things when there are not enough ports available.

Also need the 8mb flash ESP32.

If it's a bit higher in price I think it would be well worth it with these changes.

Thanks,

Jeffery

jaseiffert, 2022-12-16:
This isn't a support forum. It's meant to report problems with their code.

I will answer this time but you should find a support forum like esp32.com or others to post your questions.

I only know two things to try. Pressing the reset button while it's trying to connect to upload your code or if you have a device connected to the I2C pins, to disconnect it so it gets an error and allow you to connect to upload your code.

Thanks,

Jeffery

iwoodcock1952, 2022-12-16:
Hello Jeffery,
Thanks for your reply. I tried both if yhe suggestionswith no success.
Incidentally* Both* of my lilygo boards are dead now, neither will function
with USB TTL dongle or a clean 12.5v supply.

The board is 100% what I wanted, but seems not to be suitable after all.
Thank you for assisting me in this case. I shall return to a system based
on separate components.

Regards

Ian

jaseiffert, 2022-12-16:
Hi Ian, I'm sorry your having trouble. I looked at the schematic and the reset is connected to the EN pin on the ESP32.
I had a problem with my dongle not making a good connection where the USB C cable plugged into it. I had to push a bit hard so it was fully in hearing a click. There is about 2mm of silver connector showing between the two.

Best of luck,

Jeffery

iwoodcock1952, 2022-12-16:
Jeffery,
Thanks, but I’ve abandoned hope!! The two boards I bought are both quite
dead. Even putting 12v into the power input, nothing happens.
Thanks for all your advice.
Cheers
Ian

N6RDV, 2023-01-10:
> Jeffery, Thanks, but I’ve abandoned hope!! The two boards I bought are both quite dead. Even putting 12v into the power input, nothing happens. Thanks for all your advice. Cheers Ian

Hey Ian,
I had the same exact issue with board flashing until I figured out that the problem is in incorrect USB-2-TTL driver installed on my Mac (most likely one used for HAM radio flashing cables).

You need to download and install CH341SER driver off http://www.wch-ic.com/ and re-flash both boards. I was able to revive the board that used to boot loop after my shenanigans with it via incorrect driver...

iwoodcock1952, 2023-01-11:
Hi Jeffery, I apprecuiate you coming back to me.
One of my boards has responded but I have decided that the need to amend my sketches to adapt to this board is more than I am capable of. I'm not a regular programmer it is a kind of hobby. I now prefer the uses of an ESP32 in Arduino format with a 4-relay shield. The ESP32 board I am using has three individual I2C connectors making it very simple to use. I have ZERO programming or uploading issues.. It is as pity. This board and its 8-relay partner could have been a very elegant solution for my project.
Kind regards
Ian

N6RDV, 2023-01-11:
Hi Ian,

I'm not Jeffrey. Just came across your discussion with him and chimed in

Using an eSP32 devkit + relay shield sounds reasonable if you need extra pins.

For my project T-relay works perfectly. It's a garage/watering/lights controller (it happens that my direct burial 4ch watering wire comes into garage right next to the garage door rail, so I can use the same controller for the front yard watering and low voltage lights, to detect garage door state (closed, opening, open, closing) and to open/close the door.

Basically I connected following sensors/circuits to it:
1. 0.96" OLED display via I2C protocol
2. 371LM Garage Door Remote PCB (shorted momentary button, GND and +3.3V from one of GPIO pins are powering it instead of a 2032 battery to operate the opener)
3. Reed switch to detect whether the door is fully closed (GPIO to GND pin)
4. DIY magnetic rotary encoder to detect direction/speed of rotation and position of a counterbalancing shaft above the door (if it rotates CW - the door is opening, CCW - closing; by counting pulses I can figure out position of the door)
5. 4 x 24VAC lines connected to 4 relays (one for watering valve, other for adding a low voltage led lights along the walkway - design in progress)

Initially I planned to use Raspberry Pi Pico W on a breakout board with screw terminals + 4ch relay board side by side, but I didn't like the layout. Such arrangement required too much space. Then I came across T-relay and was able to downsize  3D printed enclosure significantly (to 8.5h of printing from 12h)

iwoodcock1952, 2023-01-11:
Hi,
I was interested to use the Lilygo board on a simple irrigation project.
Getting the relays to operate is ok now, but interfacing temp/humidity
sensor, RTC, and a 2004 lcd via custom I2C has defeated me.

I'm OK with esp32 and separate relay board, especially the KEYESTUDIO
ESP32plus board and a relay shield. Everything is default and is quite
small - Arduino Uno size. I have two such set up in my dev room on trial.

Cheers
Ian

lewisxhe, 2023-02-25:
Hello everyone, I'm lewis, I have redesigned the T-Relay, using esp32s3, isolating the drive power from the core power supply, reserving the conventional OLED and TFT interfaces, increasing the RTC clock, reserving the DS18B20 position, not soldering by default, and the driver uses shift register 595, which makes it possible to expand any number of relays, below is the product picture I designed. New discussions on the new design are welcome
[image]

iwoodcock1952, 2023-02-25:
Looks good. You have removed some of the issues I was experiencing (dedicated I2C pins etc). How is uploading - still via special dongle?
Cheers
IanW

lewisxhe, 2023-02-25:
The default upload mode is ESP32S3 USB. In order to prevent USB failure, RX TX is reserved for external USB 2 TTL download. Even if the automatic download fails, you can manually enter the download mode

jaseiffert, 2023-02-25:
I'm glad to see 5V is available.

I would like to see a power light, if you don't have one already.

Reset pins for an external reset button, not for programming just reset.

What RTC chip are you planning to use?

The DS18B20 should either have a jumper or something to disable it or make it a solder in component. Not everyone would need or want this.

Not to sure about the 20 pin connector, headers would just be fine and could be a solder in yourself or preinstalled.

The VIN on that 20 pin, is that the 7v to 28v? If so that would seem dangerous to me someone accidentally connecting that much voltage to one of their devices.

On the end of the board by CH6 relay you have VCC, would that be 5v or 3.3v ? Other than VIN you have 5v or 3v3 everywhere else.

I'm going to use my LILYGO 4 relay board for a sprinkler or drip system controller. I could see your board being use a lot for this too.

Maybe keep it to 4 relays and have an add-on board with 4 more setup on it with a connector with your 595 shift leveler ports to connect to the add-on board as well as power. An add-on would be better so you could stack one on top of the other to save room instead of one long board with 8 which might make it hard finding a good enclosure for it or making one.

On bringing out the pins you might think of making a 4 pin header with just the SDA, SCL, 3v3,  GND for I2C devices. Maybe 6 pins and add 5v and another GND to the mix. Someone then could use a individual pins to Grove connector to connect with Grove modules or any of the other systems like Gravity and what ever Adafruit is using.

And maybe a set of header pins just for SPI Connections all in one spot.

Just some thoughts, 

On my LILYGO T-Relay I'm using a Grove I2C buss board to run several things using one set of I2C pins on the LILYGO board.

I know you have to balance to what people would want and not make it too expensive and size and many other factors on designing a board but just having the 5V is a great start as I've heard many wanting to add Character displays 2x16 or 4x20 but most all of them need 5v for the back light and possibly the circuit for it work.

And It's fantastic your asking the community what they think. I wish more companies would.

X-Ryl669, 2023-02-25:
The pinout for the 2.54mm to-be-soldered header (below push buttons) seems correct. Maybe you can make it use the same order for I2C (3.3V / GND / SCL / SDA) for both the left and right part. 

The pinout for the top right connector (male) prevent using it as independent 4-pins connector. Would it be possible to swap the `GND | GND GND 5V 5V VIN VIN` to `GND | GND 5V VIN VIN 5V GND` so one can make a `5V/SDA/SCL/GND` as a 2x2 connector. However, I'm not sure the SCL / SDA on this connector is level shifted to 5V's IO, so if it's not the case, then forget this request, since one must make a intermediary PCB to level shift the IO from 3.3V to 5V anyway.

Also, I don't get the use of the CR lithium cell, since there no RTC, right ?

Finally, maybe you can add a do not populate SMD shunt resistor (like 0.075 or 0.05 ohm) on the relay's line (between the relay and the connector) used to measure the current that's consumed by the relay's load (it's always good to know what is the consumption and it doesn't cost anything except for PCB space for the shunt resistor and the 2 pins). This avoid using an external shunt resistor.

lewisxhe, 2023-02-26:
@jaseiffert 1. The power indicator uses CH7 and CH8 channels of the shift register, which are located in the middle of the circuit board
2. External RTC uses PCF8563
3. The DS18B20 is not installed by default, but is reserved
4. 20Pin Vin is 7~28V, which is used for additional boards
5. The VCC at the end of the circuit is 3v3. By default, these pins can be connected to the TFT display, because their order is the same as that of TFT
6. There are some reasons for the design of 6, because if the design is that the internal space of 4 is too small to put these components down, and the external expansion relay board has been designed, which is connected through the 20Pin seat in front of the circuit board

lewisxhe, 2023-02-26:
1. The board uses 3.3V level by default, and most matching sensors are compatible
2. At present, there is no plan to connect shunt resistance in series, and additional components and circuits are needed to measure the load consumption current
@X-Ryl669

jaseiffert, 2023-02-26:
> 

On 1, I'm not sure what you mean exactly, Can CH7 and CH8 be used in your program so you could use it to indicate power or a indicator for say the board is connected to WiFi?

2. If it is an external does the battery on the board supply power to something else? Most RTC boards I've seen have their own battery. Would there be a place to put Pins to fit one of these external boards or just use the I2C pins already there? 
3. Great!
4. Ah, that makes much better sense that the connector if for additional boards and why it has Vin. Sounds Great.
5. Ah, when you said TFT I realized why those pins were lined up the way on the edge of the board. That's Great.
6. Yes, that makes sense. Sounds like you did a lot of thinking about this design and what would be useful.

Thanks for all the great information.

lewisxhe, 2023-02-27:
1. Yes, it can be used as an indicator light
2. CR1202 is only used as the backup power supply of RTC
@jaseiffert

lewisxhe, 2023-02-28:
The board has on-board RTC, which is located next to the logo of Lilygo. 3v3 * is a separate LDO, which allows external connection of 500mA load.

lewisxhe, 2023-02-28:
Under test…[image]
[image]

jaseiffert, 2023-02-28:
> The board has on-board RTC, which is located next to the logo of Lilygo. 3v3 * is a separate LDO, which allows external connection of 500mA load.

That's great!

The pictures are great. I really like your base and cover design. Looks like the Main relay board is DIN mountable and I'm guessing the Relay modules are too but can't really see.

It looks like a great setup! Although I wouldn't need that many relays but I'm sure there are others out there that would.

The daisy chain is a great idea!

The only thing that I think your really missing is an external antenna connection for the WiFi.
It definitely needs one weather you have it in a plastic box but especially in a metal box and it could be a bit far away so you could add a different size antenna for the distance.

Either way you should sell them with an external antenna connection.

Lately for my projects I've found having external antenna on the ESP32's is a great help as I've run into WiFi problems with the built-in ones and you get a greater range. I'm not sure if they have a dual function ESP32S3 that can do both but if not I'd go with the external antenna option. The cable and Antenna are cheap for a short one.

Looking great!!

Thanks,

Jeffery

lewisxhe, 2023-03-01:
I agree with you. I have considered the antenna problem you said. We may use the ESP32S3 module with IPEX antenna, so that users can connect their own external WiFi antenna, which is suitable for the occasions installed in the iron case