Xinyuan-LilyGO/LilyGo-T-Relay #10: Will you offer these on LilyGO Amazon store?

zorrobyte, 2022-07-17:
I just bought 6 of the 4 relay boards. Will you sell the 8 relay boards on your AZ store also? I do not like ordering from Aliexpress due to shipping time.

LilyGO, 2022-07-18:
Hello. Which country are you buying from?

zorrobyte, 2022-07-18:
> Hello. Which country are you buying from?

Hello,

I am from USA! I was able to order 6x 4 relay modules with Prime shipping (2 day).

I think also if you list your relay module on esphome, you will sell many more units: https://esphome.io

Wiring up relays manually for a 3.3v board is time consuming. There is much value in your product for the US market for Home Assistant

Misiu, 2022-07-25:
@LilyGO yes, a step-by-step tutorial would be excellent.
Maybe consider adding it to the devices section on the esphome.io website (https://esphome.io/#devices).
Take a look at sonoff_4ch (https://esphome.io/devices/sonoff_4ch.html). A similar section for your products would help a lot if someone searches for info before or after buying your boards.
I'm also searching for info if ESPHome will work in T_relay 8 board.

zorrobyte, 2022-07-25:
> [Amazon Purchase link](https://www.amazon.com/dp/B0B6V3CMNB?ref=myi_title_dp).
> Are you suggesting putting the product on the Esphome? And some tutorials?

Thank you!

Note that delivery is more than a month currently, getting some available for fast delivery like the T-Relay 4 would be nice 
[image]

> @LilyGO yes, a step-by-step tutorial would be excellent.
> Maybe consider adding it to the devices section on the esphome.io website (https://esphome.io/#devices).
> Take a look at sonoff_4ch [...]

I agree that doing this would help Lilygo sell more :) There are no other relay boards I am aware of.

Yes @Misiu , they work with ESPHome, I have mine running and work just fine. Just set the relays to the documented GPIO pins on the repo. Couldn't find them on the ESPHome wiki, would have bought them sooner if I could find them more easily :)