Xinyuan-LilyGO/LilyGo-T-Relay #18: Configure the pins for I2C

rogeriopedroso, 2023-01-16:
Hello everybody,
I need to configure I2C, and I can't do it using the wire.begin(26, 14) method. Can someone explain to me how I configure the pins for I2C

trothe, 2023-02-17:
I ran into the same issue and managed to make it work like this:

```c
#include <Wire.h>

#define I2C_SDA 33
#define I2C_SCL 32

void setup() {
  Wire.begin(I2C_SDA, I2C_SCL);
}
```

Here the full example for reading a BME280:

```c
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

#define I2C_SDA 33
#define I2C_SCL 32

Adafruit_BME280 bme;

void setup() {

  Wire.begin(I2C_SDA, I2C_SCL);

  Serial.begin(9600);
  // while (!Serial);

  if (!bme.begin(0x76)) {
    Serial.println("Error, check I2c address and wiring!");
    while (1);
  }
}

void loop() {
  Serial.print("Temp:");
  Serial.print(bme.readTemperature());
  Serial.println("*C");

  Serial.print("Pressure:");
  Serial.print(bme.readPressure() / 100.0F);
  Serial.println("hPa");

  Serial.print("Humidity:");
  Serial.print(bme.readHumidity());
  Serial.println("%");

  Serial.println();
  delay(1000);
}

```

supermarioprof, 2023-08-03:
> I ran into the same issue and managed to make it work like this:
>
> ```c
> #include <Wire.h>
>
> #define I2C_SDA 33
> #define I2C_SCL 32
>
> void setup() {
>   Wire.begin(I2C_SDA, I2C_SCL);
> }
> ```
>
> Here the full [...]

How do you physically connect to pin 32 and 33? soldering directly to the ESP32 board on the T-relay board?
and, how can you use this pins for I2C whyle using for controlling relays? 
on T-relay 8X pins 32 and 33 are connected to relays K1 and K2

TAMILOLI, 2024-02-03:
Hi, We are use it **T_Relay4_ESP32** with **I2C**. In our development is IDF. we set it as pin into **SDA - 26, SCL - 14.** But Not Working as to LCD Display. Please Give me solution. In our code reference is [](https://esp32tutorials.com/i2c-lcd-esp32-esp-idf/)

trothe, 2024-02-03:
> Hi, We are use it **T_Relay4_ESP32** with **I2C**. In our development is IDF. we set it as pin into **SDA - 26, SCL - 14.** But Not Working as to LCD Display. Please Give me solution. In our code reference is https://esp32tutorials.com/i2c-lcd-esp32-esp-idf/

In my case on the T_Relay4 it was SDA 33 SCL 32. Maybe the T-Relay 8x is using other pins. you should give it a try.