Xinyuan-LilyGO/LilyGo-T-Relay #19: ESP Easy will do a hard (config) reset when connected via USB-C w/o Serial adapter

KaiDK, 2023-05-09:
Hi all,

fyi - maybe it sould be documented somewhere what we found out here: https://www.letscontrolit.com/forum/viewtopic.php?p=64424#p64424

Summary;
- ESP Easy resets configuration when RX/TX is connected on boot
- When connecting AC wall adapter via USB-C to the board while not using Lilygo Serial adapter this seems to be the case -> ESP Easy resets to config defaults

TD-er, 2023-05-09:
That ESPEasy is resetting to factory defaults when it detects a short between RX and TX is of course a feature we could disable in specific builds.
But this is a bit more serious as all data sent is thus echo'ed back to the RX.

In ESPEasy received texts are being parsed as if they are commands, as it is essentially a console.
Thus it will continuously echo the given command with some text that it was an unknown command.
But this will never stop as this error is then also echo'ed back etc.

I can imagine lots of other projects will probably not act nicely when RX and TX are shorted.

KaiDK, 2023-05-09:
Hardware solution - Use this USB cable: https://www.amazon.de/dp/B09479KTC9/
(not tested yet)