Xinyuan-LilyGO/LilyGo-T-Relay #21: Installing tasmota in T-RelayS3

Nachorov027, 2023-08-12:
Hi, I'm trying to install the full tasmota firmware in this device. I've compiled tasmota source with the #define USE_SHIFT595 and I have my board flashed and runnig but now, I'm stuck in how to configure Tasmota to the proper pins of TRelayS3 

Tasmota Signals are labeled :
74x595  SRCLK
74x595 RCLK
74x595 OE
74595  SER

How do this signal match with GPIOS in your board? 

Any clue in how to be abel to have this board (and extension) up and running?

Thx!!!

lewisxhe, 2023-08-12:
I have not used tasmota, the CH595 of Relays3 is bound to the relay, you need to drive the relay, you only need to drive the CH595

Nachorov027, 2023-08-14:
I finally get it working. These are the steps in case someone needs it. I'll also submit this info to the blacadder website. 

First, you need to compila tasmota with these options: 

`#define USE_SHIFT595
#define USE_RTC_CHIPS               // Enable RTC chip support and NTP server
#define USE_DS3231                  // [I2cDriver26] Enable DS3231 RTC (I2C address 0x68) (+1k2 code)
#define USE_RTC_ADDR    0x51`

My recomendation, use tasmocompiler docker: You simple create a new docker 

`docker pull benzino77/tasmocompiler

docker run --rm --name tasmocompiler -p 3000:3000 benzino77/tasmocompiler`
point your browser http://<you_broker_host>:3000  

It will first ask you to update to the latest tasmota code. If done it with V13, but its always good to go with the latest version (if it works :)) 
Click Next and it will ask you to enter your wifi settings. If you leave this blank, you'll get the regular Tasmota-xxxx ap to configure the relay board. 
Click next and you'll get a screen full of options. You MUST select ESP32 S3 in the second block and "humidity and temp sensors" (in order to enable I2C support and to be able to use the DS18B20 port available in TRELAY-S3,
Click nect and youl get a "Personal Config Parameters" screen, copy and paste these: 
`#define USE_SHIFT595
#define USE_RTC_CHIPS               // Enable RTC chip support and NTP server
#define USE_DS3231                  // [I2cDriver26] Enable DS3231 RTC (I2C address 0x68) (+1k2 code)
#define USE_RTC_ADDR    0x51`
Click next, and select tastmota version and languaje and finally, click "Compile"

It will take a few minutes, but when it finished, you'll find foyr buttons, click "Download Firmware_factory.bin" (VERY IMPORTAN, use the factory version of the .bin file!) 

Once you have it downloaded, flash it on your RELAY with  (you will have to install esptool if you don't have it already)
`esptool write_flash 0x0  firmwate.factory.bin`

To put your device in flashmode, just push the reset button (the one on the right), keep holding reset and push the boot mode button (left one), release reset and release boot. Then issue the previous command. 

Once flashing finish, reset again your relay board and configure wifi the usual tasmota way (unless you have already place the wifi data at compile time). 

FINAL STEPS! 

Go to Tasmota web in your device, click in "Configuration", and then click on "Extra Config". Once in this menu paste in the "Template" textbox the follwing json: 
`{"NAME":"LilyGo T-RealyS3","GPIO":[8352,1,1,1,1,8288,8320,8384,1,1,1,1,1,1,1,1,640,608,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}`
and click, "Activate" 
Go back to the main menu and voila! you have 8 relays. Relay 1-6 are the relays at the board, 7 and 8 are the red and green led. 

If you have an extension board, you need to do one more thing to enable it. 
Go to consoles menu, the first option (Console) and type this command:
`Shift595DeviceCount 2` 
Number 2 means that you have 2 devices (main+extensions). If you have 2 extension, put 3, and so. on... 

Hope it helps someone, it took me several hours to make it!! 
Regards

lewisxhe, 2023-08-14:
Good job.

diegozalez, 2024-09-07:
Hi there is a problem with the new revision boards 
[image]
As you can see enable the pin of the HT74HC595ARZ I pull high and it makes it really frustrating because if you follow all your steps it looks like everything is working but it is not.

I did a quick fix where I set GPIO 4 as a relay and if you turn it on it disables the relays and low to activate the output.

Do you suggest another way and maybe modify the template?

Thanks for your time it helps a lot.

lewisxhe, 2024-09-09:
New version When booting, you need to set GPIO4 to low to drive CH595

diegozalez, 2024-09-10:
I've set GIPO4 to low on the device configuration do you recommend a better way to do this?