Xinyuan-LilyGO/LilyGo-T-Relay #22: T-Relay S3 Chain in ESP Home

Misiu, 2023-08-17:
T-Relay S3 supports up to 3 expansion boars to have up to 24 relays.
I'd like to ask for an example with an expansion board, so it will be easier for beginners (like me) to get started.

Also, do I need additional hardware (T-U2T) to connect the board to PC?
Sadly I can't find any information and I want to buy all parts that are needed at once.

Misiu, 2023-10-10:
@LilyGO ping :)

lewisxhe, 2023-12-25:
T-Relay S3 is based on ESP32S3 and does not require an additional downloader. I think it has already been mentioned in README that programming can be done directly by inserting USB
https://github.com/xinyuan-lilygo/lilygo-t-relay/blob/main/docs/relay_esp32s3.md

lewisxhe, 2023-12-25:
The expansion board does not require special examples, only the number of relays in the sketch needs to be changed

Misiu, 2023-12-30:
@lewisxhe thank you for the reply,
I'm not sure if I need to add another hub or should I use the same hub and use other switches.
According to https://esphome.io/components/sn74hc595.html one SN74HC595 can be used with 8 pins, if you want to use more you must daisy chain them, but I don't know how to configure that in ESPHome.
Guess I'll wait for the board to be available on my local amazon to be able to order it and rest everything with option to return the board if something doesn't work.

fgastald, 2024-02-10:
same problem here cannot find adeguate informations for configure correctly the YAML file.

szab0lcs, 2024-04-01:
Here is my working code in ESPHome.
The `sr_count` is define how many chains is daisy chained together.
```yaml
esphome:
  name: test-6-relay
  friendly_name: Test 6 Relay

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

sn74hc595:
  - id: "sn74hc595_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 2

# Individual outputs
switch:
  - platform: gpio
    # SN74HC595 Pin #0
    name: "Switch 1"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 0
      number: 0
      inverted: false
      id: relay1

  - platform: gpio
    # SN74HC595 Pin #1
    name: "Switch 2"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 1
      number: 1
      inverted: false
      id: relay2

  - platform: gpio
    # SN74HC595 Pin #2
    name: "Switch 3"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 2
      number: 2
      inverted: false
      id: relay3

  - platform: gpio
    # SN74HC595 Pin #3
    name: "Switch 4"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 3
      number: 3
      inverted: false
      id: relay4

  - platform: gpio
    # SN74HC595 Pin #4
    name: "Switch 5"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 4
      number: 4
      inverted: false
      id: relay5

  - platform: gpio
    # SN74HC595 Pin #5
    name: "Switch 6"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 5
      number: 5
      inverted: false
      id: relay6

  - platform: gpio
    # SN74HC595 Pin #8 chain: 1
    name: "Switch Chain_1"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 8
      number: 8
      inverted: false
      id: relay1_chain

  - platform: gpio
    # SN74HC595 Pin #9 chain: 2
    name: "Switch Chain_2"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 9
      number: 9
      inverted: false
      id: relay2_chain

  - platform: gpio
    # SN74HC595 Pin #10 chain: 3
    name: "Switch Chain_3"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 10
      number: 10
      inverted: false
      id: relay3_chain

  - platform: gpio
    # SN74HC595 Pin #11 chain: 4
    name: "Switch Chain_4"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 11
      number: 11
      inverted: false
      id: relay4_chain

  - platform: gpio
    # SN74HC595 Pin #12 chain: 5
    name: "Switch Chain_5"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 12
      number: 12
      inverted: false
      id: relay5_chain

  - platform: gpio
    # SN74HC595 Pin #13 chain: 6
    name: "Switch Chain_6"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 13
      number: 13
      inverted: false
      id: relay6_chain

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

ota:
  password: "XXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-6-Relay Fallback Hotspot"
    password: "ZBnPdRLZ3foz"

captive_portal:
```

Misiu, 2024-04-02:
@szab0lcs thank you for sharing your code.
Did you have flickering issues - https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/issues/24#issuecomment-1868701351
When you restart the device does any of the repays turn on/off? I want to use T-Relay S3 as a garage controller to control doors and read a couple of DS18B20 sensors.

szab0lcs, 2024-04-02:
> @szab0lcs thank you for sharing your code.
>
> Did you have flickering issues - https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/issues/24#issuecomment-1868701351
>
> When you restart the device does any [...]

Yes I have but I will try to solve that issue. If I get any solution I'll keep you posted.

szab0lcs, 2024-04-03:
> > @szab0lcs thank you for sharing your code.
> > Did you have flickering issues - [#24 (comment)](https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/issues/24#issuecomment-1868701351)
> > When you restart the device does any of the repays turn on/off? I want to use T-Relay S3 as a garage controller to control doors and read a couple of DS18B20 sensors.
> 
> Yes I have but I will try to solve that issue. If I get any solution I'll keep you posted.

https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/issues/24#issuecomment-2034384008

cremer42, 2024-09-25:
> Here is my working code in ESPHome. The `sr_count` is define how many chains is daisy chained together.
>
> ```yaml
> esphome:
>   name: test-6-relay
>   friendly_name: Test 6 Relay
>
> esp32:
>   board: [...]

@szab0lcs I basically copied your yaml, loaded it to my board I got today and I cannot get relays to even function.... Am I missing something?