Xinyuan-LilyGO/LilyGo-T-Relay #23: ESPHome example of T-Relay 5V 8 Channel Relay and T-Relay W5500 Shield

Misiu, 2023-10-10:
First of all, thank you for creating this repository with sample configurations.
I'd like to ask for additional config for T-Relay with W5500 Shield and ESPHome.

Thank you :)

lewisxhe, 2023-12-25:
I searched for Ephome and currently do not have any components for w5500

B48D81EFCC, 2024-07-25:
Hi,

ESPHome supports W5500 now.

bartoszkaminski, 2025-01-21:
Hi! I'm also trying to get T-Relay 8 channel to work with T-Relay W5500 Shield using ESPHome. Do you by any change have any working example?

lewisxhe, 2025-01-22:
Since I don't have actual hardware to test, you could try testing this

```c
esphome:
  name: t-relay-8

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# wifi:
#   ssid: "ssid"
#   password: "password"

#   # Enable fallback hotspot (captive portal) in case wifi connection fails
#   ap:
#     ssid: "T-Relay Fallback Hotspot"
#     password: "g8TaLY13NVW6"

ethernet:
  type: W5500
  clk_pin: GPIO22
  mosi_pin: GPIO26
  miso_pin: GPIO34
  cs_pin: GPIO27
  interrupt_pin: GPIO36
  reset_pin: GPIO23

switch:
  - platform: gpio
    pin: GPIO33
    name: "Relay #1"
    id: relay1

  - platform: gpio
    pin: GPIO32
    name: "Relay #2"
    id: relay2

  - platform: gpio
    pin: GPIO13
    name: "Relay #3"
    id: relay3

  - platform: gpio
    pin: GPIO12
    name: "Relay #4"
    id: relay4

  - platform: gpio
    pin: GPIO21
    name: "Relay #5"
    id: relay5

  - platform: gpio
    pin: GPIO19
    name: "Relay #6"
    id: relay6

  - platform: gpio
    pin: GPIO18
    name: "Relay #7"
    id: relay7

  - platform: gpio
    pin: GPIO5
    name: "Relay #8"
    id: relay8

captive_portal:
```

Llgok, 2025-10-30:
@Misiu @bartoszkaminski @Mattims 
Hello, please try the latest .yaml file [T-Relay-8-ESP32-W5500.yaml](https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/ESPHome/T-Relay-8-ESP32-W5500.yaml). If it loads correctly, you will obtain the following serial log：

```bash
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6276
load:0x40078000,len:15736
load:0x40080400,len:4
load:0x40080404,len:3860
entry 0x40080634
I (29) boot: ESP-IDF 5.4.2 2nd stage bootloader
I (29) boot: compile time Oct 29 2025 10:14:56
I (30) boot: Multicore bootloader
I (31) boot: chip revision: v3.1
I (33) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (64) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (71) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (77) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (84) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=15350h ( 86864) map
I (131) esp_image: segment 1: paddr=00025378 vaddr=3ff80000 size=0001ch (    28) load
I (131) esp_image: segment 2: paddr=0002539c vaddr=3ffb0000 size=02540h (  9536) load
I (138) esp_image: segment 3: paddr=000278e4 vaddr=40080000 size=08734h ( 34612) load
I (156) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=44f34h (282420) map
I (252) esp_image: segment 5: paddr=00074f5c vaddr=40088734 size=061c4h ( 25028) load
I (269) boot: Loaded app from partition at offset 0x10000
I (269) boot: Disabling RNG early entropy source...
[0;32m[I][logger:121]: Log initialized[0m
[0;35m[C][safe_mode:084]: Unsuccessful boot attempts: 3[0m
[0;36m[D][esp32.preferences:149]: Writing 1 items: 0 cached, 1 written, 0 failed[0m
[0;32m[I][app:073]: Running through setup()[0m
[0;35m[C][component:173]: Setup preferences took 0ms[0m
[0;36m[D][switch:024]: 'Relay #1' Turning OFF.[0m
[0;36m[D][switch:063]: 'Relay #1': Sending state OFF[0m
[0;36m[D][switch:024]: 'Relay #1' Turning OFF.[0m
[0;35m[C][component:173]: Setup gpio.switch took 7ms[0m
[0;36m[D][switch:024]: 'Relay #2' Turning OFF.[0m
[0;36m[D][switch:063]: 'Relay #2': Sending state OFF[0m
[0;36m[D][switch:024]: 'Relay #2' Turning OFF.[0m
[0;35m[C][component:173]: Setup gpio.switch took 20ms[0m
[0;36m[D][switch:024]: 'Relay #3' Turning OFF.[0m
[0;36m[D][switch:063]: 'Relay #3': Sending state OFF[0m
[0;36m[D][switch:024]: 'Relay #3' Turning OFF.[0m
[0;35m[C][component:173]: Setup gpio.switch took 20ms[0m
[0;36m[D][switch:024]: 'Relay #4' Turning OFF.[0m
[0;36m[D][switch:063]: 'Relay #4': Sending state OFF[0m
[0;36m[D][switch:024]: 'Relay #4' Turning OFF.[0m
[0;35m[C][component:173]: Setup gpio.switch took 20ms[0m
[0;36m[D][switch:024]: 'Relay #5' Turning OFF.[0m
[0;36m[D][switch:063]: 'Relay #5': Sending state OFF[0m
[0;36m[D][switch:024]: 'Relay #5' Turning OFF.[0m
[0;35m[C][component:173]: Setup gpio.switch took 10ms[0m
[0;36m[D][switch:024]: 'Relay #6' Turning OFF.[0m
[0;36m[D][switch:063]: 'Relay #6': Sending state OFF[0m
[0;36m[D][switch:024]: 'Relay #6' Turning OFF.[0m
[0;35m[C][component:173]: Setup gpio.switch took 11ms[0m
[0;36m[D][switch:024]: 'Relay #7' Turning OFF.[0m
[0;36m[D][switch:063]: 'Relay #7': Sending state OFF[0m
[... 67 lines cut ...]
  Link Speed: 100[0m
[0;35m[C][ethernet:387]:   CLK Pin: 22
  MISO Pin: 34
  MOSI Pin: 26
  CS Pin: 27[0m
[0;35m[C][ethernet:399]:   IRQ Pin: 36[0m
[0;35m[C][ethernet:401]:   Reset Pin: 23
  Clock Speed: 8 MHz[0m
[0;35m[C][ethernet:416]:   Type: W5500[0m
[0;35m[C][esphome.ota:093]: Over-The-Air updates:
  Address: test.local:3232
  Version: 2[0m
[0;35m[C][safe_mode:018]: Safe Mode:
  Successful after: 60s
  Invoke after: 10 attempts
  Duration: 300s[0m
[0;33m[W][safe_mode:030]: Last reset too quick; invoke in 7 restarts[0m
[0;35m[C][api:222]: Server:
  Address: test.local:6053
  Listen backlog: 4
  Max connections: 8[0m
[0;35m[C][api:234]:   Noise encryption: NO[0m
[0;35m[C][mdns:179]: mDNS:
  Hostname: test[0m
```