Xinyuan-LilyGO/LilyGo-T-Relay #24: Prevent relay trigger on boot or reset with esphome code

davidusb-geek, 2023-12-10:
I'm using a very similar code to the one here for esphome and Home Assistant.
Everything works fine except that when the board resets or reboot all the relay are triggered.
Very inconvenient when controlling a garage door for example.
Any suggestions to solve this and avoid these relay triggering?
I have tried some solutions using the relay options on esphome (ALWAYS_OFF, etc.) but without success.
Any help will be much appreciated.

Misiu, 2023-12-18:
@davidusb-geek what model or relay are you using (T-Relay, T-Relay 8, or T-Relay S3)?
I'm thinking about ordering those relays and I'd like to know did you solved the issue you described.
I also want to use it to control garage doors, so I can't allow random opening.
Thank you for the reply.

davidusb-geek, 2023-12-18:
I'm using the t-relay 8.
The issue is not solved, it's open.
So I disconnected the garage door for now.

Misiu, 2023-12-18:
Sad to hear that.
Mayby try to contact support via website chat.

davidusb-geek, 2023-12-23:
Oh man, I've just solved this after months looking for a solution!
So the thing is that I've already tested with options `mode: INPUT_PULLUP` and `inverted: True` but without good results. But my problem was with indentation in my yaml for esphome. These parameters were just being ignored.
Today I just found that the correct way to pass this parameters and make my switch work without triggering at boot is this:

First define the output pins:
```
output:
  - platform: gpio
    pin: 
      number: GPIO33
      mode: INPUT_PULLUP
      inverted: True
    id: output_pin_1 # garage door
```

Then I define my switch for the garage door like this:
```
switch:
  - platform: output
    restore_mode: ALWAYS_OFF
    id: garage_door_bare_switch
    output: output_pin_1
  - platform: template
    restore_mode: ALWAYS_OFF
    name: "Garage Door Switch"
    icon: "mdi:garage"
    id: garage_door_bare_template_switch
    lambda: 'return id(garage_door_bare_switch).state;'
    turn_on_action:
      - switch.turn_on: garage_door_bare_switch
      - delay: 0.5s
      - switch.turn_off: garage_door_bare_switch
    turn_off_action:
      - switch.turn_on: garage_door_bare_switch
      - delay: 0.5s
      - switch.turn_off: garage_door_bare_switch
```

And that's it, works like charm.

Misiu, 2023-12-24:
@davidusb-geek thank you for sharing the solution :)
Now I think I order the boards with confidence.
Did you maybe attach the board to a DIN rail? I was searching for a mounting solution and I would like to know what you choose.

davidusb-geek, 2023-12-24:
Wait! I spoke too soon!
Actually the switch is not working. I thought it was fine because it didn't trigger at boot and it was showing fine on Home Assistant (and it was very late yesterday night) . But it is not working, the switch is not responding. My guess is that the INPUT_PULLUP is not supported by the hardware.
I'm reopening the issue :-(

davidusb-geek, 2023-12-24:
Yes I'm using DIN rail. I'll post the aliexpress link for that piece. It is a shame that I have this issue because the board quality is very good.

davidusb-geek, 2023-12-24:
These DIN supports work fine: Just found this amazing item on AliExpress. Check it out! 0,34€ | 1 Pair DRG-01 PCB Mounting Brackets Screws Green For DIN 35 Mounting Rails Adapter Replacements Parts
https://a.aliexpress.com/_EwgujuV

lewisxhe, 2023-12-25:
Relay4/realy8 should not have this problem, but there is indeed a chance that this problem will occur on realy6

davidusb-geek, 2023-12-25:
Ok I will try the INPUT_PULLUP option on relay4 or relay8 and see how it goes.

davidusb-geek, 2023-12-30:
> Relay4/realy8 should not have this problem, but there is indeed a chance that this problem will occur on realy6

I just tested it on relay 4 (ie. GPIO21), still doesn't works.

Misiu, 2023-12-30:
> Relay4/realy8 should not have this problem, but there is indeed a chance that this problem will occur on realy6

@lewisxhe could you please say more about relay6 (T-Relay S3)? I need to be 100% sure that the relays won't switch on/off after power loss. I want to control my garage door, so if this issue occurs my garage will open!

davidusb-geek, 2023-12-31:
> Relay4/realy8 should not have this problem, but there is indeed a chance that this problem will occur on realy6

So I completely misinterpreted this reply.

Misiu, 2023-12-31:
> > Relay4/realy8 should not have this problem, but there is indeed a chance that this problem will occur on realy6
> 
> So I completely misinterpreted this reply.

I'm not sure if I understand it right. I'm assuming it was related to product types and not to relay in T-Relay 8

davidusb-geek, 2024-01-01:
Ok, got it working now! Tested it for real and everything is working fine and as expected.

I tested many options and finally got it working with this configutaion:
```
output:
  - platform: gpio
    pin: 
      number: GPIO12
      inverted: True
      mode:
        pullup: True
        output: True
    id: output_pin_1 # garage door
```

The switch part of the yaml code is the same that I posted earlier in this issue.

Cheers!!!

Misiu, 2024-01-01:
@davidusb-geek thank you for sharing a solution. Yesterday I ordered a T-relay 8 board, it should arrive in about 3-4 weeks so I'll also test all the options.

szab0lcs, 2024-04-03:
I got it working with an easier method, the `mode: OUTPUT` was the key part:

```yaml
sn74hc595:
  - id: "sn74hc595_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 2

# Individual outputs
switch:
  - platform: gpio
    # SN74HC595 Pin #0
    name: "Switch 1"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 0
      number: 0
      inverted: false
      mode: OUTPUT
    id: relay1
```

forgotmyprozac, 2024-11-17:
> Ok, got it working now! Tested it for real and everything is working fine and as expected.
>
> I tested many options and finally got it working with this configutaion:
>
> ```
> output:
>   - platform: [...]

Hey mate, I really like this board but the relays triggering on boot is killing me. What's the issue and how did you deal with it? I'm using ESPHome on the Lilygo T-Relay with Home Assistant. Thanks a bunch.

davidusb-geek, 2024-11-17:
> @davidusb-geek
> 
> Hey mate, I really like this board but the relays triggering on boot is killing me. What's the issue and how did you deal with it? I'm using ESPHome on the Lilygo T-Relay with Home Assistant. Thanks a bunch.

Sorry to hear that. I also liked this board.
At first that solution that I posted worked and then some of these flickering came back.
I had enough with that so I changed to another board. I changed for the Kincony A4, working like charm, no more flickering problems.

Of course that requires changing the board. So probably try the solution proposed by @szab0lcs

lewisxhe, 2024-11-18:
@forgotmyprozac  Can you please tell me what version of your board is? Can you take a picture of the front of the board for me to see?

forgotmyprozac, 2024-11-18:
@lewisxhe

> @forgotmyprozac Can you please tell me what version of your board is? Can you take a picture of the front of the board for me to see?

Images below. Thank you.

https://i.ibb.co/Hzg5BXc/PXL-20241118-064935332.jpg
https://i.ibb.co/19ZnpMT/PXL-20241118-064929583.jpg
https://i.ibb.co/h1c5ypK/PXL-20241118-064924938.jpg

lewisxhe, 2024-11-18:
This is an early version and can be solved by adding a resistor, see the picture
[image]
@forgotmyprozac

lewisxhe, 2024-11-18:
After this change, 74HC595 will be enabled through GPIO4, and only GPIO4 low level can drive the relay

forgotmyprozac, 2024-11-18:
@lewisxhe 
@davidusb-geek 

> After this change, 74HC595 will be enabled through GPIO4, and only GPIO4 low level can drive the relay

Thank you very much for the help. What wattage for the resistor and what AWG wire for IO4?

What do you mean "after normal power on, set IO4 to LOW" > Do I need to set this in my code? So it will boot up and be high, then the code will pull it low, then relays can be used?

How can we tell what version the board is? I have another one coming from Aliexpress and want to know if it will have the same problem.

https://www.aliexpress.com/item/1005005576226160.html?spm=a2g0o.order_list.order_list_main.5.2e521802TrkVnZ

Thank you

lewisxhe, 2025-04-01:
The new version has a resistor here, and the version number is V1.1

[image]

[image]

Misiu, 2025-04-01:
Thank you for such a fast reply. Now I need to contact local sellers and ask for the board version. The rev must be 1.1 for the main board or also for the chain boards?
P.S. Thank you for the W5500 examples :)

lewisxhe, 2025-04-01:
Chain boards have not been updated, they are still the same as before, it should not affect