Xinyuan-LilyGO/LilyGo-T-Relay #26: Internal PCF8563 not incrementing with T-Relay S3 [H629]

Olli69, 2024-01-31:
Hi, 
I tested internal PCF8563 rtc with the sample program RTC_PCF8563_TimeSynchronization. It got updates from ntp-server and set date/time to the rtc, but the rtc was not incrementing. For example just after ntp update loop() printed out 
```
Hardware clock :Wednesday, January 31 2024 17:25:17
System   clock :Wednesday, January 31 2024 17:25:17
````
but 10 seconds later:
```

Hardware clock :Wednesday, January 31 2024 17:25:17
System   clock :Wednesday, January 31 2024 17:25:27
```

I had the same problem with Adafruit library as well as https://github.com/Makuna/Rtc library. Could it be a hardware problem? I just got the board from LilyGo (order 13992).

lewisxhe, 2024-01-31:
Can you use I2C scan to test whether PCF8563 is online?
Normally, as long as the PCF8563 is powered on, the time will accumulate.

Olli69, 2024-01-31:
> Can you use I2C scan to test whether PCF8563 is online? Normally, as long as the PCF8563 is powered on, the time will accumulate.

It has a fresh CR-1220 battery (does it save the static time but not accumulate?) . Also response from i2c scan (`WIRE.begin(16,17);`): `I2C device found at address 0x51  !`

lewisxhe, 2024-01-31:
Then I may suspect that the RTC crystal is not working properly. Is it possible for you to measure the clock waveform of 32.768khz?

Olli69, 2024-01-31:
I do have an oscilloscope, but not very professional with it. If you could send  a pic of pins to measure, I could try.

lewisxhe, 2024-01-31:
I will send it to you tomorrow and add instructions

lewisxhe, 2024-02-01:
[image]

Check these two pins, adjust the oscilloscope probe to the 10X gear, then adjust the period to 10us and the voltage to 50mV. If your oscilloscope supports period measurement, you can open the period measurement option and observe whether there is a 32.768khz waveform output.

Olli69, 2024-02-01:
I got quite nice sine waves from pin 1 and pin 2.  But are the voltage levels correct for PCF8563 to operate, OSCI between 356mV-828mV? I measured VDD voltage levels from PIN 8: 2.88V-3.12V. 

[image]
[image]

lewisxhe, 2024-02-01:
There seems to be no problem. The crystal is already oscillating. You can try my test file to see if it accumulates every one second.

There seems to be no problem. The crystal is already oscillating. You can try my test file to see if it accumulates every one second.

[RTC_Firmware.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/files/14121926/RTC_Firmware.zip)

How to flash firmware , you can see here, the method is the same, the writing address is 0x0

https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/firmware/README.MD#2%EF%B8%8F%E2%83%A3how-to-flash-

Olli69, 2024-02-01:
Thanks for additional diagnostics. I flashed your test file and it does not seem to accumulate.
```
Hardware clock :Monday, January 01 2024 00:00:00
Hardware clock :Monday, January 01 2024 00:00:00
Hardware clock :Monday, January 01 2024 00:00:00
Hardware clock :Monday, January 01 2024 00:00:00
Hardware clock :Monday, January 01 2024 00:00:00
Hardware clock :Monday, January 01 2024 00:00:00
Hardware clock :Monday, January 01 2024 00:00:00
```

lewisxhe, 2024-02-01:
@Olli69 Sorry for confusing you
At present, it seems from all aspects that the PCF8563 chip may be damaged, causing it to not work properly. Please contact customer service for after-sales service.

Olli69, 2024-02-01:
@lewisxhe , no problem! Thank you for prompt responses with diagnostics. So good that we got this far so quickly. I have contacted your sales and asked for a replacement board.

lewisxhe, 2024-02-01:
You're welcome. If you don't have any other questions, please close it.

closed 2024-02-01 by Olli69