Xinyuan-LilyGO/LilyGo-T-Relay #33: External pin setted with input pull-up is not working.

DAZABAMUKER, 2024-05-12:
I connected external pins with switches like images below.

At first, the switches worked well.
But now the external pins float(intermittently triggered)

how to fix it?

[image]

```
#include <PubSubClient.h>
#include <WiFi.h>

const char *ssid =  "########";   
const char *password = "#######"; 

const char* ID = "Outdoor Lights";  // Name of our device, must be unique
const char* LightTOPIC = "outdoor/light/sub";
const char* INTOPIC = "Outdoor/light/pub";

const char* mqttUser = "########";
const char* mqttPassword = "########";
const char* broker = "########";

WiFiClient wclient;
PubSubClient client(wclient); 

char messages[50];
String msg = "";
int light_sel = 0;
String light_val = "";
int lightPin_num[7] = {33, 32, 13, 12, 21, 19, 18};
int switch_num[7] = {2, 23, 26, 27, 15, 14, 4};
boolean light_num[7] = {false, false, false, false, false, false, false};
boolean light_state[7] = {true, true, true, true, true, true, true};
boolean light_state_old[7] = {true, true, true, true, true, true, true};
const char* LightTOPICs[7] = {
  "outdoor/light/entrance_front",
  "outdoor/light/entrance_back",
  "outdoor/light/agung-i",
  "outdoor/light/outdoor1",
  "outdoor/light/outdoor2",
  "outdoor/light/outdoor3",
  "outdoor/light/outdoor4"
};

void setup_wifi() {
  Serial.print("\nConnecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password); 
  while (WiFi.status() != WL_CONNECTED) { 
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect(ID, mqttUser, mqttPassword)) {
      Serial.println("connected");
      Serial.print("Publishing to: ");
      Serial.println(LightTOPIC);
      Serial.println('\n');
      client.subscribe(INTOPIC);

    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println("\n try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Received messages: ");
  Serial.print(INTOPIC);
  msg = "";
  light_sel = 0;
  light_val = "";
  for(int i=0; i<length; i++){
    //Serial.println((char) payload[i]);
    msg += (char)payload[i];
  }
  if (msg == "none") {
    return;
  }
  Serial.print(" ");
  Serial.println(String(msg));
  int msg_index = msg.indexOf('_');
  light_sel = msg.substring(0, msg_index).toInt();
  light_val = msg.substring(msg_index+1, msg.length());
  Serial.println(light_sel);
  Serial.println(light_val);
  boolean light = false;
  if (light_val == "f") {
    light = false;
  } else if (light_val == "t") {
    light = true;
  }
  if (light_num[light_sel] != light) {
    lightControl(light_sel);
  }
  delay(100);
}

void lightControl(int selected) {
  Serial.println("clicked");
  light_num[selected] = !light_num[selected];
  digitalWrite(lightPin_num[selected], light_num[selected]);
  snprintf(messages, 75, "%s", String(light_num[selected]));
  client.publish(LightTOPICs[selected], messages);
}

void setup() {
  // put your setup code here, to run once:
  for(int i = 0; i < 7; i++){
    pinMode(lightPin_num[i], OUTPUT);
    pinMode(switch_num[i], INPUT_PULLUP);
  }
  dacWrite(25, 100);

  Serial.begin(115200);
  setup_wifi(); // Connect to network
  client.setServer(broker, 1883);
  client.setCallback(callback);
}

void loop() {
  // put your main code here, to run repeatedly:
  if (!client.connected()){
    reconnect();
  }
  client.loop();

  for(int i = 0; i < 7; i++) {
    light_state[i] = digitalRead(switch_num[i]);
    if (light_state[i] != light_state_old[i]) {
      lightControl(i);
      light_state_old[i] = light_state[i];
    }
  }
  delay(100);
}
```

lewisxhe, 2024-05-25:
Adding a 10K pull-up resistor to the GPIO can solve the floating problem, and can also be set as an input pull-up, but only with GPIO below GPIO33.
```
pinMode(gpio, INPUT_PULLUP);
```

DAZABAMUKER, 2024-05-25:
Thanks I understood.

closed 2024-07-10 by Llgok