Xinyuan-LilyGO/LilyGo-T-Relay #35: Movement and ghost switching

charlesnoble, 2024-05-16:
In our motorhome we have:

- 1 * T-Relay 4 Channel
- 2 * T-Relay S3 (both with an extension board

This bug is hard to describe, but I will do my best here. The two sets of T-Relay S3 relays (each a master and slave unit) are mounted via two of the four holes on the corners of the board.

I'm running all boards via Home Assistant / ESPHome. Here is my code for one of the boards:

```
esphome:
  name: "esp32-behind-freezer"

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ##REMOVED##

ota:
  password: ##REMOVED##

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Behind-Freezer Fallback Hotspot"
    password: ##REMOVED##

captive_portal:

sn74hc595:
  - id: "sn74hc595_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 2

# Individual outputs
switch:
  - platform: gpio
    # SN74HC595 Pin #0
    name: "Behind Freezer Switch 1"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 0
      number: 0
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_1

  - platform: gpio
    # SN74HC595 Pin #1
    name: "Behind Freezer Switch 2"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 1
      number: 1
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_2

  - platform: gpio
    # SN74HC595 Pin #2
    name: "Behind Freezer Switch 3"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 2
      number: 2
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_3

  - platform: gpio
    # SN74HC595 Pin #3
    name: "Behind Freezer Switch 4"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 3
      number: 3
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_4

  - platform: gpio
    # SN74HC595 Pin #4
    name: "Behind Freezer Switch 5"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 4
      number: 4
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_5

  - platform: gpio
    # SN74HC595 Pin #5
    name: "Behind Freezer Switch 6"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 5
      number: 5
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_6

  - platform: gpio
    # SN74HC595 Pin #8 chain: 1
    name: "Behind Freezer Switch 7"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 8
      number: 8
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_7

  - platform: gpio
    # SN74HC595 Pin #9 chain: 2
    name: "Behind Freezer Switch 8"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 9
      number: 9
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_8

  - platform: gpio
    # SN74HC595 Pin #10 chain: 3
    name: "Behind Freezer Switch 9"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 10
      number: 10
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_9

  - platform: gpio
    # SN74HC595 Pin #11 chain: 4
    name: "Behind Freezer Switch 10"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 11
      number: 11
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_10

  - platform: gpio
    # SN74HC595 Pin #12 chain: 5
    name: "Behind Freezer Switch 11"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 12
      number: 12
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_11

  - platform: gpio
    # SN74HC595 Pin #13 chain: 6
    name: "Behind Freezer Switch 12"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 13
      number: 13
      mode: OUTPUT
      inverted: false
      id: behind_freezer_switch_12
```

Everything was working smoothly until one day, I got up from the seat above which one of the relay board(s) was mounted, and it immediately triggered all relays to turn off. It's super strange, as nothing from the seat touches the board or relays. Now it happens every other time from getting up from the seat.

Here is what the relay board looks like when a relay is normally triggered (putting our stairs up and down): 

https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/assets/972843/2232f1f9-238c-4752-8eee-aac9da434deb

And here is what the relay does when I hope up from the seat directly above it: 

https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/assets/972843/cb347f5c-0d15-4c06-a544-172deec671d4

You'll notice the relay board doesn't move. However, all relays turn off, and a green light labeled 'CH8' switches on. I then switch the stairs back on via Home Assistant. This causes the green light to turn off and the relay to turn on again.

In some cases, other relays on the board switch on with the green LED going on. It's not until I trigger any relay via Home Assistant that things return to normal.

The pattern of the behavior is erratic and hard to measure. It doesn't happen every time I hop up from the seat. This issue also occurs with the other set of T-Relay S3 boards under the bed. However, in this case, it's even harder to capture the occurrence of this issue. 

I'm sorry I can't give you too much more information. The ESPhome logs do not mention any issues.

lewisxhe, 2024-05-25:
I don't understand, you mean the relay will activate randomly?

Can you remove the expansion relay cable first and then test it?

charlesnoble, 2024-05-25:
> I don't understand, you mean the relay will activate randomly?

That's correct, my friend. The movement seems to trigger it. You'll see in the second video that the the relay is 'on', and when I hop up from my seat (which is directly above the relay board), the relay turns 'off', and that green 'ch8' light turns on. I can't see any noticeable pressure on the board itself. It's just the general movement of the motorhome when I get up from the seat.

When I open home assistant to check, nothing has changed on home assistant. Home assistant still thinks the relay is 'on'. So, on Home Assistant, I turn the relay 'off' and 'on' again to return things to normal.

 > Can you remove the expansion relay cable first and then test it?

OK, I can test this, but it's hard to replicate this issue as it's not consistent.

lewisxhe, 2024-05-25:
I have been using T-Relay-S3 for three months and have not found any problems with it automatically triggering. However, I placed it on the table. You can try removing the extension cable first and see if it will cause any problems when used alone. this situation

marc-gist, 2024-06-03:
Sounds like a power issue to me. ensure its receiving adequate power supply, no lose connections. check esphome and see if the board is rebooting during this event. if so, you have a power issue, or the board is damaged and a trace on the pcb has come lose/broken.