Xinyuan-LilyGO/LilyGo-T-Relay #37: Can't switch T-Relay status

JpEncausse, 2024-08-22:
Hello,
I bought a T-Relay S3 I try yo make it work, my code seems ok but nothing happened and I don't know how to debug.

My code is inspired from HT74HC595 here is the Setup part:

```
ShiftRegister74HC595_NonTemplate *HT74HC595 = NULL;
void setupRelay()
{
    HT74HC595 = new ShiftRegister74HC595_NonTemplate(8, HT74HC595_DATA, HT74HC595_CLOCK, HT74HC595_LATCH);

    pinMode(HT74HC595_OUT_EN, OUTPUT);
    digitalWrite(HT74HC595_OUT_EN, HIGH); // Close output data HT74HC595

    digitalWrite(HT74HC595_OUT_EN, LOW);
    HT74HC595->setAllLow();
    digitalWrite(HT74HC595_OUT_EN, HIGH);
}
```

and the loop part

void loopRelay()
{
    static unsigned long relay_t0 = millis();
    unsigned long t1 = millis();
    int delta = t1 - relay_t0;
    
    for (int i = 0; i < relayCount; i++) {
        
        if (relayTime[i] == 0 && relayState[i] == false){ continue; }
        relayTime[i] -= delta;

        if (relayTime[i] > 0){
            if (relayState[i] == false){
                digitalWrite(HT74HC595_OUT_EN, LOW);
                HT74HC595->set(i, HIGH, true);
                digitalWrite(HT74HC595_OUT_EN, HIGH);
                relayState[i] = true;
                Serial.printf("Relay %d: %d\n", i, relayTime[i]);
            }
            continue;
        }

        relayTime[i]  = 0;
        relayState[i] = false;
        digitalWrite(HT74HC595_OUT_EN, LOW);
        HT74HC595->set(i, LOW, true);
        digitalWrite(HT74HC595_OUT_EN, HIGH);
        Serial.printf("Relay %d: OFF\n", i);
    }
    relay_t0 = t1;
}

I have a WebSocket that change the relayTime so every x millisecond watch and decide to keep the relay HIGH or LOW.
- The Serial.printf() provide the right information.
- The wire are on COM (Ground) and NO (10V) if I make them touch physically it also works
So when I put, let say, 2000ms in item 3 it should set relay 3 to HIGH for 2000ms then LOW.

But nothing happened and my code seems OK.
Can you please tell me what I'm missing ?

lewisxhe, 2024-08-23:
Please refer to [UnitTestExample](https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/ac5a099de31cd6af825681d2ca6bdb7cf5708ff8/examples/T-Relay-ESP32S3/UnitTestExample/UnitTestExample.ino#L109) , HT74HC595_OUT_EN only needs to be set to LOW after power-on. Do not operate this GPIO afterwards.

```
ShiftRegister74HC595_NonTemplate *HT74HC595 = NULL;
void setupRelay()
{
    HT74HC595 = new ShiftRegister74HC595_NonTemplate(8, HT74HC595_DATA, HT74HC595_CLOCK, HT74HC595_LATCH);

    pinMode(HT74HC595_OUT_EN, OUTPUT);
    digitalWrite(HT74HC595_OUT_EN, HIGH); // Close output data HT74HC595

    HT74HC595->setAllLow();
    digitalWrite(HT74HC595_OUT_EN, LOW);

}

void loopRelay()
{
    static unsigned long relay_t0 = millis();
    unsigned long t1 = millis();
    int delta = t1 - relay_t0;
    
    for (int i = 0; i < relayCount; i++) {
        
        if (relayTime[i] == 0 && relayState[i] == false){ continue; }
        relayTime[i] -= delta;

        if (relayTime[i] > 0){
            if (relayState[i] == false){
                HT74HC595->set(i, HIGH, true);
                relayState[i] = true;
                Serial.printf("Relay %d: %d\n", i, relayTime[i]);
            }
            continue;
        }

        relayTime[i]  = 0;
        relayState[i] = false;
        HT74HC595->set(i, LOW, true);
        Serial.printf("Relay %d: OFF\n", i);
    }
    relay_t0 = t1;
}

```

JpEncausse, 2024-08-23:
Ohhh thanks it works !
I looked into multiple sample doing this so I was not sure. Many thanks !

closed 2024-08-23 by lewisxhe