Xinyuan-LilyGO/LilyGo-T-Relay #38: Pin Mapping for T-Relay with 4 Relays

JohannesNeu, 2024-09-14:
Hello everyone,

I bought a https://www.lilygo.cc/products/t-relay with 4 Relays and flashed tasmota, which works very well.

Next I would like to attach two to three MAX31865 via SPI and I am therefore looking for the correct PINs.
According to this schematic: https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/Schematic/T_Relay4_ESP32.pdf
(page one upper right corner) The proper PINs on the header would be:

- Header PIN 5 = GPIO PIN 14 [IO14] = CLK
- Header PIN 7 = GPIO PIN 15 [IO15] = MOSI
- Header PIN 6 = GPIO PIN 2 [IO2] = MISO

The schematics describe a "ESP32-WROVER-**B**".
While the ESP on my board is labeled "ESP32-WROVER-**E**"

According to the ESP32-WROVER-E datasheet from Espressif: https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf (page 5 - 7) the PINs would be different (table 4).

- Header PIN 5 = GPIO PIN 14 [IO14] = HSPICLK = CLK
- Header PIN 15 = GPIO PIN 39 [IO39] = HSPID = MOSI
- Header PIN 9 = GPIO PIN 12 [IO12] = HSPIQ = MISO

The IO15, which is used as MOSI in the lilygo schematics is the HSPICS0 chip select according to the Espressif documentation and IO2 is HSPIWP according to it.

I would appreciate your help to figure out which configuration is correct and if both are feasible an explanation where this difference comes from.

kind regards,
Johannes

lewisxhe, 2024-09-18:
No, you just need to specify which GPIO to use
```
int sck = Less than GPIO 33;
int miso = any_pin;
int mosi = Less than GPIO 33;
SPI.begin(sck,miso,mosi);

```

JohannesNeu, 2024-09-23:
Thank you for the reply, but what you are describing is software SPI, right?
I tried it with tasmota and if I want to use SSPI it works with every pin.
But what if i want to use hardware SPI ?

lewisxhe, 2024-09-24:
This is hardware SPI