Xinyuan-LilyGO/LilyGo-T-Relay #40: Not able to get relays working

cremer42, 2024-09-25:
Hello, I purchased this board due to many reasons. This is supposed to replace a generic 4 channel ESP device I already have built/programmed.

I have put what I would consider a basic ESPhome program on it and I am not able to get relays to pull in. Am I missing something? Any help would be appreciated.

Below is my current code:

substitutions:
  name: "Lily Test"
  friendly_name: lily-test
  relay_1_name: Switch 1
  relay_2_name: Switch 2
  relay_3_name: Switch 3
  relay_4_name: Switch 4
  relay_5_name: Switch 5
  relay_6_name: Switch 6
  dallas_1: Pre Heater Temp
  dallas_2: Post Heater Temp
  dallas_3: Panel Temperature
  

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: Lilygo T-Relay S3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  platform: esphome
  password: !secret ota_pw

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${friendly_name}
    password: !secret wifi_password

web_server:
  port: 80      

sn74hc595:
  - id: "relay_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 2

# Individual outputs
switch:
  - platform: gpio
    # SN74HC595 Pin #0
    name: ${relay_1_name}
    pin:
      sn74hc595: relay_hub
      number: 0  # Use pin number 0
      inverted: false
      id: relay_1

  - platform: gpio
    # SN74HC595 Pin #1
    name: ${relay_2_name}
    pin:
      sn74hc595: relay_hub
      number: 1  # Use pin number 1
      inverted: false
      id: relay_2

  - platform: gpio
    # SN74HC595 Pin #2
    name: ${relay_3_name}
    pin:
      sn74hc595: relay_hub
      number: 2  # Use pin number 2
      inverted: false
      id: relay_3

  - platform: gpio
    # SN74HC595 Pin #3
    name: ${relay_4_name}
    pin:
      sn74hc595: relay_hub
      number: 3  # Use pin number 3
      inverted: false
      id: relay_4

  - platform: gpio
    # SN74HC595 Pin #4
    name: ${relay_5_name}
    pin:
      sn74hc595: relay_hub
      number: 4  # Use pin number 4
      inverted: false
      id: relay_5

  - platform: gpio
    # SN74HC595 Pin #5
    name: ${relay_6_name}
    pin:
      sn74hc595: relay_hub
      number: 5        # Use pin number 5
      inverted: false
      id: relay_6

captive_portal:

cremer42, 2024-09-25:
```
substitutions:
  name: "pool-control-v2"
  friendly_name: Pool Controller v.2.0
  relay_1_name: Switch 1
  relay_2_name: Switch 2
  relay_3_name: Switch 3
  relay_4_name: Switch 4
  relay_5_name: Switch 5
  relay_6_name: Switch 6
  dallas_1: Pre Heater Temp
  dallas_2: Post Heater Temp
  dallas_3: Panel Temperature
  

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: Lilygo T-Relay S3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret pool_controller_api_key

ota:
  platform: esphome
  password: !secret pool_controller_ota_pw

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${friendly_name}
    password: !secret wifi_password

web_server:
  port: 80      

sn74hc595:
  - id: "relay_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 2

# Individual outputs
switch:
  - platform: gpio
    # SN74HC595 Pin #0
    name: ${relay_1_name}
    pin:
      sn74hc595: relay_hub
      number: 0  # Use pin number 0
      inverted: false
      id: relay_1

  - platform: gpio
    # SN74HC595 Pin #1
    name: ${relay_2_name}
    pin:
      sn74hc595: relay_hub
      number: 1  # Use pin number 1
      inverted: false
      id: relay_2

  - platform: gpio
    # SN74HC595 Pin #2
    name: ${relay_3_name}
    pin:
      sn74hc595: relay_hub
      number: 2  # Use pin number 2
      inverted: false
      id: relay_3

  - platform: gpio
    # SN74HC595 Pin #3
    name: ${relay_4_name}
    pin:
      sn74hc595: relay_hub
      number: 3  # Use pin number 3
      inverted: false
      id: relay_4

  - platform: gpio
    # SN74HC595 Pin #4
    name: ${relay_5_name}
    pin:
      sn74hc595: relay_hub
      number: 4  # Use pin number 4
      inverted: false
      id: relay_5

  - platform: gpio
    # SN74HC595 Pin #5
    name: ${relay_6_name}
    pin:
      sn74hc595: relay_hub
      number: 5        # Use pin number 5
      inverted: false
      id: relay_6

captive_portal:
```

closed 2024-09-25 by cremer42

jasonmhite, 2024-10-01:
For anybody coming across this and wondering why ESPHome is struggling with this device, @cremer42 's solution is correct, you need to pull up the enable pin on the shift register, which is tied to GPIO4. This is not mentioned anywhere in the docs [page](https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/docs/RELAY_ESP32S3.MD) for this board. @LilyGO or @lewisxhe should probably mention this in the docs.

A slightly improved version of @cremer42 's solution that gets the polarity of the enable pin right and sets it to an internal switch so that it doesn't show up in Home Assistant:

```yaml
switch:
  - platform: gpio
    pin: 
      number: GPIO4
      inverted: True
    name: "SN74HC595 Enable"
    internal: True
    id: ch_en_pin
    restore_mode: ALWAYS_ON
```

cremer42, 2024-10-02:
> For anybody coming across this and wondering why ESPHome is struggling with this device, @cremer42 's solution is correct, you need to pull up the enable pin on the shift register, which is tied to [...]

You are correct. I wanted mine to show up in HA in my circumstance. Mostly for troubleshooting. This was actually a benefit for me. Since I've automated my pool using this board, I have an automation that locks out my equipment. My automation can now just trigger GPI04 and turn all the relays off. 

But yes, if this was better documented, it would've made my life a lot easier.

lewisxhe, 2024-10-04:
Hi @cremer42 

I have set the initial voltage of HC595 4 Pin to LOW in yaml, so it should work normally?
https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/fd145e8e2d1f830eabd84213349a2f8e72d09321/ESPHome/T-Relay-ESP32S3.yaml#L20