Xinyuan-LilyGO/LilyGo-T-Relay #41: I2C and DS18B20 Pin Definition

cremer42, 2024-09-25:
I've looked and looked at the schematics page.... Does someone have a better pin breakdown of what the GPIO's are they relate to?

I am trying to setup a one_wire bus and a i2c bus using ESPhome. 

I plant to solder wires to the DS18B20 location on the board, so knowing the built in pin would be beneficial. I was also considering using the Qwiic / Stemma connector for my I2C bus, so knowing what Pins that uses would be greatly appreciated as well.

Thank you,!

lewisxhe, 2024-09-26:
Do you mean T-Relay-S3? If yes, look here

https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/fd145e8e2d1f830eabd84213349a2f8e72d09321/examples/T-Relay-ESP32S3/DS18x20_Temperature/DS18x20_Temperature.ino#L10

cremer42, 2024-09-26:
Ok... Thank you. What about the I2C? Is it defined? Specifically the Qwiic connector? If not, can it be changed to say, GPIO 16 and 17?

lewisxhe, 2024-09-27:
DS18B20 PCB reserves a 4.7k resistor position, close to DS18B20
You should check if the delay is caused by the network delay, not the relay driver delay.

cremer42, 2024-09-27:
Sorry for the inexperienced reply. But how do I check both of those? 

Thanks

cremer42, 2024-09-27:
Thanks. I'll see if I can figure out how to do that into esphome.

cremer42, 2024-09-28:
I have no clue how to "write" this driver relay.... All I know is when I insruct a relay to turn on/off, i have a 10-18 second delay. Any help you can provide would be appreciated.

cremer42, 2024-09-28:
Unsure if this could be causing issues....

[23:27:17][E][PCF8563:025]: Communication with PCF8563 failed!
[23:27:17][E][component:082]:   Component pcf8563.time is marked FAILED

lewisxhe, 2024-10-01:
https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/firmware%2FLilyGo-T-Relay-S3%2Ffirmware_LilyGo-Relay-UnitTest_20240409.bin

Try this factory test, using io0 to switch the relay.

cremer42, 2024-10-04:
Maybe I am a little fresh when it comes to these things... I installed the BIN, but how do I connect to it to switch on io0?

lewisxhe, 2024-10-04:
Just uploaded a firmware for relay testing, you can use this to test
https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/firmware/LilyGo-T-Relay-S3/firmware_relay_test_241004.bin