Xinyuan-LilyGO/LilyGo-T-Relay #44: S3relay ESPHome - Output ch_en_pin error

dew1989, 2024-10-17:
It seems that ESPHome will not compile with your posted code due to this section 
```
output:
  - id: ch_en_pin
    pin: 4  # enable 74ch595  
    initial_state: low
```

ESPHome requires a "platform" to be set. Please advise on the correct code

lewisxhe, 2024-10-17:
Try this

```
esphome:
  name: lilygo-relays3
  build_path: .esphome/build/lilygo-relays3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

sn74hc595:
  - id: "sn74hc595_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 1

output:
  - platform: gpio
    id: ch_en_pin
    pin: 4

# Individual outputs
switch:
  - platform: gpio
    name: "Switch 1"
    pin:
      sn74hc595: sn74hc595_hub
      number: 0
      inverted: false
      id: relay1

  - platform: gpio
    name: "Switch 2"
    pin:
      sn74hc595: sn74hc595_hub
      number: 1
      inverted: false
      id: relay2

  - platform: gpio
    name: "Switch 3"
    pin:
      sn74hc595: sn74hc595_hub
      number: 2
      inverted: false
      id: relay3

  - platform: gpio
    name: "Switch 4"
    pin:
      sn74hc595: sn74hc595_hub
      number: 3
      inverted: false
      id: relay4

  - platform: gpio
    name: "Switch 5"
    pin:
      sn74hc595: sn74hc595_hub
      number: 4
      inverted: false
      id: relay5

  - platform: gpio
    name: "Switch 6"
    pin:
      sn74hc595: sn74hc595_hub
      number: 5
      inverted: false
      id: relay6
mqtt:
  broker: Your MQTT server
  # username: Your MQTT Username
  # password:  Your MQTT Password
  port: Your MQTT port
  
logger:
  level: DEBUG
  baud_rate: 115200
  logs:
    mqtt.component: DEBUG
    mqtt.client: ERROR

api:
  encryption:
    key: "E4AvftHULay+f6wwnKgHzsu7q7nzUdcrHp/Mjhyyjvk="

ota:
  password: "0204c885b8f04777ae94a8e51b4619c5"

wifi:
  ssid: "Your WiFi ssid"
  password: "Your WiFi password"

  ap:
    ssid: "Relay Fallback Hotspot"
    password: "fltzKdbsbDEn"

captive_portal:

```

lewisxhe, 2024-10-18:
Sorry, I found a mistake, I have updated esphome, please retest https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/ESPHome/T-Relay-ESP32S3.yaml

closed 2024-10-18 by dew1989