Xinyuan-LilyGO/LilyGo-T-Relay #45: relay S3 no wifi and strange behaviar

bkbartk, 2024-10-17:
Hello,

I just  got my Lilygo T-RelayS3, but I can't get it to work as expected.
I flashed esphome on it using the config below
wifi settings are in my `.base.yaml`
I connected the external antenna,
-still my device does not connect to wifi.
-no AP is created for fall back
-on boot, random switches are enabled, no clue way.
-when I try to get anything out of the serial logger the serial adapter keeps disconnecting which makes logging impossible.
So actually I don't get anything to work.

I power my device using usb-c, but powering is from my PC or using an RPI-4 adapter doesn't matter, the behaviar is the same.

I must be missing something but can't figure what it is.
```
esphome:
  name: lilygotest
  friendly_name: lilygotest

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

packages:
  base: !include .base.yaml

sn74hc595:
  - id: "sn74hc595_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 1

# output:
#   - id: ch_en_pin
#     pin: 4  # enable 74ch595  
#     initial_state: low

# Individual outputs
switch:
  - platform: gpio
    # SN74HC595 Pin #0
    name: "Switch 1"
    restore_mode: RESTORE_DEFAULT_OFF 
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 0
      number: 0
      inverted: false
      id: relay1
      mode: output

  - platform: gpio
    # SN74HC595 Pin #1
    name: "Switch 2"
    restore_mode: RESTORE_DEFAULT_OFF 
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 1
      number: 1
      inverted: false
      id: relay2
      mode: output

  - platform: gpio
    # SN74HC595 Pin #2
    name: "Switch 3"
    restore_mode: RESTORE_DEFAULT_OFF 
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 2
      number: 2
      inverted: false
      id: relay3
      mode: output

  - platform: gpio
    # SN74HC595 Pin #3
    name: "Switch 4"
    restore_mode: RESTORE_DEFAULT_OFF 
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 3
      number: 3
      inverted: false
      id: relay4
      mode: output

  - platform: gpio
    # SN74HC595 Pin #4
    name: "Switch 5"
    restore_mode: RESTORE_DEFAULT_OFF 
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 4
      number: 4
      inverted: false
      id: relay5
      mode: output

  - platform: gpio
    # SN74HC595 Pin #5
    name: "Switch 6"
    restore_mode: RESTORE_DEFAULT_OFF 
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 5
      number: 5
      inverted: false
      id: relay6
      mode: output

```

lewisxhe, 2024-10-18:
Sorry, I found a mistake, I have updated esphome, please retest https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/ESPHome/T-Relay-ESP32S3.yaml

bkbartk, 2024-10-18:
I applied the changes for `GPIO4` but it doesn't help.
this serial addapter keeps diconnecting `USB JTAG/Serial debug urit(COM3) - paired`
so the log only says
```
[22:53:42]ESP-ROM:esp32s3-20210327
Terminal disconnected: NetworkError: The device has been lost.
```
the only way the keep the com port steady is to hold the boot button. but then I don't get logs either.
I also set the restore mode to always_On, but that doesn't change anything, It's just like my flash doesn't stick to the device.

bkbartk, 2024-10-18:
some comment, it seems to work when I use framework: arduino,
but all fails when using framework: esp-idf.
is there no support for this? and if not will there ever be?

lewisxhe, 2024-10-21:
Are you using esphome? Does it keep disconnecting? What method did you use to write to it?