Xinyuan-LilyGO/LilyGo-T-Relay #46: Relay S3 - main board not working but chained board works

dew1989, 2024-10-17:
I just got a S3 relay and one daisy chained board. I flashed via ESPHome, I have full working control over the daisy chained board, but cannot get the relays to turn on or off on the main board. Wifi connects, and I have no trouble reflashing, but cannot control the main board. Do I have a faulty board? 

I think my code is good, but here it is regardless:
```
esphome:
  name: lilygo-relay
  friendly_name: relay
  build_path: .esphome/build/lilygo-relays3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: Arduino

# Enable logging
logger:
  baud_rate: 115200

# Enable Home Assistant API
api:

ota:
  platform: esphome

wifi:
  ssid: ###
  password: ###

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  # SSID is device name. No password
  ap: {}

captive_portal:

output:
  - platform: gpio
    id: ch_en_pin
    pin: 4
    inverted: true
    
sn74hc595:
  - id: "sn74hc595_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 2

# Individual outputs
switch:
  - platform: gpio
    # SN74HC595 Pin #0
    name: "Switch 1"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 0
      number: 0
      inverted: false
      id: relay1

  - platform: gpio
    # SN74HC595 Pin #1
    name: "Switch 2"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 1
      number: 1
      inverted: false
      mode: OUTPUT
      id: relay2

  - platform: gpio
    # SN74HC595 Pin #2
    name: "Switch 3"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 2
      number: 2
      inverted: false
      mode: OUTPUT
      id: relay3

  - platform: gpio
    # SN74HC595 Pin #3
    name: "Switch 4"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 3
      number: 3
      inverted: false
      mode: OUTPUT
      id: relay4

  - platform: gpio
    # SN74HC595 Pin #4
    name: "Switch 5"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 4
      number: 4
      inverted: false
      mode: OUTPUT
      id: relay5

  - platform: gpio
    # SN74HC595 Pin #5
    name: "Switch 6"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 5
      number: 5
      inverted: false
      mode: OUTPUT
      id: relay6

  - platform: gpio
    # SN74HC595 Pin #8 chain: 1
    name: "Switch Chain_1"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 8
      number: 8
      inverted: false
      mode: OUTPUT
      id: relay1_chain

  - platform: gpio
    # SN74HC595 Pin #9 chain: 2
    name: "Switch Chain_2"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 9
      number: 9
      inverted: false
      mode: OUTPUT
      id: relay2_chain

  - platform: gpio
    # SN74HC595 Pin #10 chain: 3
    name: "Switch Chain_3"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 10
      number: 10
      inverted: false
      mode: OUTPUT
      id: relay3_chain

  - platform: gpio
    # SN74HC595 Pin #11 chain: 4
    name: "Switch Chain_4"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 11
      number: 11
      inverted: false
      mode: OUTPUT
      id: relay4_chain

  - platform: gpio
    # SN74HC595 Pin #12 chain: 5
    name: "Switch Chain_5"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 12
      number: 12
      inverted: false
      mode: OUTPUT
      id: relay5_chain

  - platform: gpio
    # SN74HC595 Pin #13 chain: 6
    name: "Switch Chain_6"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 13
      number: 13
      inverted: false
      mode: OUTPUT
      id: relay6_chain

```

lewisxhe, 2024-10-18:
Sorry, I found a mistake, I have updated esphome, please retest https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/ESPHome/T-Relay-ESP32S3.yaml

dew1989, 2024-10-18:
This fixed it. Thank you!

closed 2024-10-18 by dew1989