Xinyuan-LilyGO/LilyGo-T-Relay #48: Replace esp module?

drosner, 2024-12-11:
Not sure if this can be answered but i have 2 boards that i fried the esp module and want to replace. I tried using an air station like i would a normal IC chip but it wouldn't budge. Anyone have any suggestions or familiar with how these are manufactured that can give me some pointers to remove the esp module from the PCB board?

lewisxhe, 2024-12-13:
You need a heat gun, the temperature needs to be adjusted to about 330°, and then it must be heated for several minutes before it can be completely disassembled.

drosner, 2024-12-13:
Thank you sir! I'm using a hot air rework station and have been using 400C. Heating from top and bottom for 20 mins! Thing won't budge....

lewisxhe, 2024-12-13:
You can add some solder oil to assist heating, or you can add some low-temperature solder paste to neutralize the lead-free solder on the PCB.
Also, the air outlet of the hot air gun should cover the entire ESP32 as much as possible.