Xinyuan-LilyGO/LilyGo-T-Relay #50: T-Relay 8 channel doesn't work with W5500 Shield attached

bartoszkaminski, 2025-01-08:
Hello!

I purchased LilyGo T-Relay 8ch and I wanted to use it with T-Relay W5500 shield to have Ethernet. I soldered all pins of the shield but now my relay board doesn’t start at all. When I supply 12V  to the relay board there’s no reaction, it doesn’t go online. Ethernet port doesn’t blink when cable is attached.

[image]
[image]
[image]
[image]

Can you help me figure out what is wrong? Before soldering the shield I had ESPHome installed on the relay board and it worked perfectly.

Llgok, 2025-01-09:
@bartoszkaminski 
Hello, thank you for your report. I noticed that you seem to have soldered the W5500 module to the bottom of the board. Doing so will cause both boards to short circuit. The correct approach is to solder the W5500 to the top of the main board, as shown in the following image.
[image]
[image]

bartoszkaminski, 2025-01-09:
I was assuming that two square pins must be connected together.

Can you confirm that my understanding of pins orientation is correct now?

[image]
[image]

BTW, do you think I fried the board? If it works after correctly connecting the shield then I'm good or will its lifespan be limited due to the fact that I shorted it?

Llgok, 2025-01-09:
@bartoszkaminski 
Yes, the connection method you are demonstrating now is correct. Could you try this method to see if the device can function properly? Previously, reversing the power supply might have caused a short circuit on the board.
[image]

Misiu, 2025-04-01:
@Llgok please consider adding a picture to the product description page showing the correct placement of the W5500 Shield (both to the W5500 shield and to the T-Relay board pages.
It would be super helpful for less technical users like myself.

Llgok, 2025-04-02:
@Misiu 
It has been updated to the [FAQ](https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay?tab=readme-ov-file#faq).

Llgok, 2025-06-23:
The issue was closed due to no response for a long time. If you still have questions, please reopen it.

closed 2025-06-23 by Llgok