Xinyuan-LilyGO/LilyGo-T-Relay #51: T-Relay Chain problems

carbuthn, 2025-01-09:
1. I could not find a schematic for the T-Relay Chain board.
2. All of the pictures show the cable that connect the T-Relay S3 motherboard to the T-Relay Chain connected to the left side of the T-Relay Chain board, this is labeled as an output on the board. I was able to get the board to work once I moved the cable to the input (right) side of the board.
3. Would like to know the pin to turn off the red LED on the T-Relay Chain board.

lewisxhe, 2025-01-10:
The red LED cannot be turned off, it is the power indicator

carbuthn, 2025-01-11:
I was afraid of that, I didn't have one on the main board, only the chain board. Time for some black paint.