Xinyuan-LilyGO/LilyGo-T-Relay #53: esp32_8relays module max current

qvlogan88, 2025-04-02:
hi all,

I would like to protect my circuit with a fuse, but I couldn't find the maximum current that flows through the board in the docs
someone knows it? thanks in advance and forgive for bad english!

qvlogan88, 2025-04-02:
nono, I'm talking about the maximum current that the card can handle before burning out

lewisxhe, 2025-04-02:
Are you referring to the board's power output? As far as I know, this board only has 3.3V pins exposed, right?

lewisxhe, 2025-04-02:
Input current of the board? Are you talking about the current limit for powering the board? There is no input current limit. It is recommended to use DC power supply. Only about 10W of current is needed to work well.