Xinyuan-LilyGO/LilyGo-T-Relay #54: Lily go ESP32-S3 T6 relay

tirty7th, 2025-04-26:
Hi. I’m new to programming and don’t have much experience, but I am having a difficult time programming the dev board for t6 relay. Do I need the usb adapter to program it properly? I have tried the most basic code to toggle on the relays but none of the pin outs numbers work. I was able to toggle on all 6 relays with a random AI generated code, but it was inconsistent and activated random relays. There is very little information available on this dev board so I am asking for help. The GitHub pages seem to be out of date and geared towards only the esp dev boards, not the esp32-S3 ones. Any assistance is greatly appreciated.

lewisxhe, 2025-04-26:
This is a register controlled using a shift register, not relays controlled individually using GPOP, have you looked at the examples here?
https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/tree/main/examples/T-Relay-ESP32S3

What code did you use for testing?

tirty7th, 2025-04-26:
Does this mean that it requires an additional board to program? I’m fairly
new to programming and so I assumed that it would use GPIO to control the
relays.

lewisxhe, 2025-04-26:
This is the most basic example of controlling a relay https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/examples/T-Relay-ESP32S3/HT74HC595/HT74HC595.ino

No additional programmer is required, please see README, all documents are up to date.

tirty7th, 2025-04-26:
I receive this error: Compilation error:
ShiftRegister74HC595_NonTemplate.h: No such file or directory

Is there a URL I need to add to the board manager? What step am I missing?

thanks.

lewisxhe, 2025-04-26:
[image]

tirty7th, 2025-04-26:
I'm on a macbook. I copied the lib files to the library folder. I now get
this error:

Compilation error: pin_config.h: No such file or directory

lewisxhe, 2025-04-26:
Isn't pinconfig.h here?

[image]

lewisxhe, 2025-04-26:
[image]

Just copy the pin definition to the sketch. Your question is really elementary. 😂😂😂

tirty7th, 2025-04-26:
[image]

lewisxhe, 2025-04-26:
I just deleted it and put it on the sketch.

tirty7th, 2025-04-26:
I mentioned that I am very inexperienced at programming. However, what is
elementary to you is not to others. The code that you referenced in your
picture is missing.

lewisxhe, 2025-04-26:
That's okay. Just do the best I can.

tirty7th, 2025-04-26:
I am confused then. If it is on the sketch then why is it saying that the
directory does not exist?

lewisxhe, 2025-04-26:
Please show me a screenshot of your ide

tirty7th, 2025-04-26:
I don't understand what you mean.

lewisxhe, 2025-04-26:
Send me a screenshot of the error interface and your sketch directory folder.

tirty7th, 2025-04-26:
[image]

lewisxhe, 2025-04-26:
[image]
Apparently your Arduino IDE doesn't open pinconfig.h, can you check if they are in the same directory?

tirty7th, 2025-04-26:
[image]

How would the pin_config.h file be present? It is not on GitHub

tirty7th, 2025-04-26:
If you have the file pin_config.h available, that would be helpful.

lewisxhe, 2025-04-26:
No need... I have updated the sketch... Just download this file again.

https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/examples/T-Relay-ESP32S3/HT74HC595/HT74HC595.ino

tirty7th, 2025-04-26:
Thank you very much for your assistance. The code is working now.