Xinyuan-LilyGO/LilyGo-T-Relay #55: T-Relay S3 unresponsive when connected to T-Relay Chain

R0X, 2025-06-15:
Hi there,

i am having problems with the T-Relay S3 when i connect one T-Relay Chain to it.
I setup the relays to switch on for 400ms and then turn off again. When i connect the T-Relay Chain they sometimes just switch on only a few ms. Furthermore sometimes random relays are activated. I am using adjustable 12v 24 Watt power supply. Rising the Voltage to 15v didn't make a difference.

In this [Video](https://youtube.com/shorts/UzArPEQac_0?feature=share) i triggered the cee2_switch switch on action

```
    turn_on_action:
      - switch.turn_on: relay2
      - delay: 400ms
      - switch.turn_off: relay2
 ```

three times, only the last one succeeded. In attemp 1 and 2 you can hear the relay click very shortly but no led indication. 

The following esphome config is used:

```
 
esphome:
  name: $devicename
  comment: ${device_description}

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

logger:

api:
  password: !secret api_password

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password
  
  manual_ip:
    static_ip: ${device_ip}
    gateway: 172.16.49.254
    subnet: 255.255.255.0
    dns1: 172.16.40.5
    dns2: 172.16.47.15

  ap:
    ssid: "${devicename} Fallback Hotspot"
    password: !secret wifi_fallback_password

time:
  - platform: homeassistant
    id: homeassistant_time

web_server:
  port: 80
  auth:
    username: admin
    password: !secret web_server_password

text_sensor:
  - platform: version
    name: "Version"
    icon: mdi:cube-outline

sn74hc595:
  - id: "sn74hc595_hub"
    data_pin: 7
    clock_pin: 5
    latch_pin: 6
    sr_count: 2

binary_sensor:
  - platform: status
    name: "Status"
    device_class: connectivity

  - platform: gpio
    id: input1
    internal: true
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: false
  - platform: gpio
    id: input2
    internal: true
    pin:
      number: GPIO21
      mode: INPUT_PULLUP
      inverted: false
  - platform: gpio
    id: input3
    internal: true
    pin:
      number: GPIO47
      mode: INPUT_PULLUP
      inverted: false
  - platform: gpio
    id: input4
    internal: true
    pin:
      number: GPIO38
      mode: INPUT_PULLUP
      inverted: false
  - platform: gpio
    id: input5
    internal: true
    pin:
      number: GPIO48
      mode: INPUT_PULLUP
      inverted: false
  - platform: gpio
    id: input6
    internal: true
    pin:
      number: GPIO45
      mode: INPUT_PULLUP
      inverted: false

sensor:
  - platform: wifi_signal
    name: "Wifi Signal"
    update_interval: 60s
    icon: mdi:wifi

  - platform: uptime
    name: "Uptime"
    update_interval: 60s
    icon: mdi:clock-outline

switch:
  - platform: gpio
    id: ch_en
    pin: 
      number: GPIO4
      inverted: true
    restore_mode: ALWAYS_ON

  - platform: gpio
    # SN74HC595 Pin #0
    name: "Switch 1"
    id: relay1
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 0
      number: 0
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #1
    name: "Switch 2"
    id: relay2
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 1
      number: 1
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #2
    name: "Switch 3"
    id: relay3
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 2
      number: 2
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #3
    name: "Switch 4"
    id: relay4
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 3
      number: 3
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #4
    name: "Switch 5"
    id: relay5
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 4
      number: 4
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #5
    name: "Switch 6"
    id: relay6
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 5
      number: 5
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #8 chain: 1
    name: "Switch Chain_1"
    id: relay1_chain
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 8
      number: 8
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #9 chain: 2
    name: "Switch Chain_2"
    id: relay2_chain
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 9
      number: 9
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #10 chain: 3
    name: "Switch Chain_3"
    id: relay3_chain
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 10
      number: 10
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #11 chain: 4
    name: "Switch Chain_4"
    id: relay4_chain
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 11
      number: 11
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #12 chain: 5
    name: "Switch Chain_5"
    id: relay5_chain
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 12
      number: 12
      inverted: false
      mode: OUTPUT

  - platform: gpio
    # SN74HC595 Pin #13 chain: 6
    name: "Switch Chain_6"
    id: relay6_chain
    internal: true
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 13
      number: 13
      inverted: false
      mode: OUTPUT

  - platform: template
    name: "CEE1"
    id: cee1_switch
    lambda: |-
      return id(input1).state;
    turn_on_action:
      - switch.turn_on: relay1
      - delay: 400ms
      - switch.turn_off: relay1
    turn_off_action:
      - switch.turn_on: relay1_chain
      - delay: 400ms
      - switch.turn_off: relay1_chain
    optimistic: false

  - platform: template
    name: "CEE2"
    id: cee2_switch
    lambda: |-
      return id(input2).state;
    turn_on_action:
      - switch.turn_on: relay2
      - delay: 400ms
      - switch.turn_off: relay2
    turn_off_action:
      - switch.turn_on: relay2_chain
      - delay: 400ms
      - switch.turn_off: relay2_chain
    optimistic: false

  - platform: template
    name: "CEE3"
    id: cee3_switch
    lambda: |-
      return id(input3).state;
    turn_on_action:
      - switch.turn_on: relay3
      - delay: 400ms
      - switch.turn_off: relay3
    turn_off_action:
      - switch.turn_on: relay3_chain
      - delay: 400ms
      - switch.turn_off: relay3_chain
    optimistic: false

  - platform: template
    name: "CEE4"
    id: cee4_switch
    lambda: |-
      return id(input4).state;
    turn_on_action:
      - switch.turn_on: relay4
      - delay: 400ms
      - switch.turn_off: relay4
    turn_off_action:
      - switch.turn_on: relay4_chain
      - delay: 400ms
      - switch.turn_off: relay4_chain
    optimistic: false

  - platform: template
    name: "CEE5"
    id: cee5_switch
    lambda: |-
      return id(input5).state;
    turn_on_action:
      - switch.turn_on: relay5
      - delay: 400ms
      - switch.turn_off: relay5
    turn_off_action:
      - switch.turn_on: relay5_chain
      - delay: 400ms
      - switch.turn_off: relay5_chain
    optimistic: false

  - platform: template
    name: "CEE6"
    id: cee6_switch
    lambda: |-
      return id(input6).state;
    turn_on_action:
      - switch.turn_on: relay6
      - delay: 400ms
      - switch.turn_off: relay6
    turn_off_action:
      - switch.turn_on: relay6_chain
      - delay: 400ms
      - switch.turn_off: relay6_chain
    optimistic: false
```

R0X, 2025-06-17:
i used the config from you but still same error
[see this Video](https://youtube.com/shorts/FPoKLlbd9nU)

first three attempt you hear the relay click but too short and the led also doesnt stay on.
only fourth attempt led stays on

when i unplug the chain it works normaly.

lewisxhe, 2025-06-18:
There is no problem with the yaml file, and the single test is OK. It seems to be a network delay problem. Connect the USB and check the control output of the serial monitor. After clicking the switch, the serial console should immediately output the controlled relay number.

lewisxhe, 2025-06-21:
Two test methods,
1. Write the relay driver test to see if it is normally closed https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/blob/main/examples/T-Relay-ESP32S3/HT74HC595/HT74HC595.ino

2. You said that the mainboard is abnormal when connected to 220V, so you can take the board outside the distribution box for testing first

From my personal opinion, it seems that the strong current causes the wifi signal to be abnormal. Have you checked the serial output message according to the method I mentioned above?