Xinyuan-LilyGO/LilyGo-T-Relay #56: Library and EthernetServer server issue

Deoyjadzo, 2025-07-08:
Hello everyone. I am trying to use the Lilygo board 4 relay with the w5500 shield and I am facing some dificulties. My guess it is a library issue, but please feel free to correct me if not.

I am using arduino IDE 2.0.2 with board set as ESP32 dev board - same setup that I use for all other Lilygo boards projects and all other projects works smoothly.

But right now I am facing this error:

C:\Users\Luiz Eduardo\Documents\Arduino\trygithublilygorj45\trygithublilygorj45.ino:135:16: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'
 EthernetServer server(80);
                ^~~~~~
In file included from C:\Users\Luiz Eduardo\Documents\Arduino\trygithublilygorj45\trygithublilygorj45.ino:118:
c:\Users\Luiz Eduardo\Documents\Arduino\libraries\Ethernet\src/Ethernet.h:254:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^~~~~~~~~~~~~~
 ^

exit status 1

Compilation error: cannot declare variable 'server' to be of abstract type 'EthernetServer'

In a basic research I found that this should be a library issue. But indeed I tryed with 5 or 6 types of ethernet libraries. Ethernet, Ethernet_Generic, WIZ, and some of EthernetWebServer by khoi. None of them worked and I got the same error for all.

So basically my first question is, which library setup is being used here ? Is there anything else I need to know to make it work ?

lewisxhe, 2025-07-09:
[image]
Which version did you use ?

Llgok, 2025-07-11:
@Deoyjadzo 
Hello, for W5500 examples in ESPLib v3.0 and above, please use this one: [w5500_esplib_v3.0](https://github.com/Xinyuan-LilyGO/LilyGo-T-Relay/tree/main/examples/T-Relay-4-ESP32/w5500_esplib_v3.0).

Deoyjadzo, 2025-07-11:
> [@Deoyjadzo](https://github.com/Deoyjadzo) Hello, may I ask which example you are using for compilation? Currently, I am using the W5500_EthernetServer_Relay4.ino from esplib-v2.0.14, and it compiles and runs normally.
> 
> <img alt="Image" width="1868" height="1507" src="https://private-user-images.githubusercontent.com/135582120/464453736-aa668b64-11b6-447c-89ce-f50874cc3b14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIyMzY5MDMsIm5iZiI6MTc1MjIzNjYwMywicGF0aCI6Ii8xMzU1ODIxMjAvNDY0NDUzNzM2LWFhNjY4YjY0LTExYjYtNDQ3Yy04OWNlLWY1MDg3NGNjM2IxNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxMVQxMjIzMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMWRkMDhhYTJjMDZlOGZjMmVlZjQ0YTMxOGRlZDUwMDg3YzA0YzdiOTczNmYwZDkwYWVmZTUyN2E2NDlkMzVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.SaJeuoqu08yDvH6DdIrl9Vdj4Mh6CKg4fMPWKCPYuKA">

Which Ethernet library did you use in this case ?